QuinnFreedman/modular #1: Issues activating options

damwan, 2021-06-28:
Hello!
I reproduced your clock and I must say that it is by far the best I have seen so far for such a simple circuit. But I have some difficulty activating options.
Activating PAUSE_BUTTON_ENABLED or TAP_TEMPO_ENABLED returns the serial message "SSD1306 allocation failed"
And selecting a screen saver other than SS_BARS causes this compilation error:

> Eurorack \ Quinn Freedman Clock \ Clock \ Clock.ino: 280: 34: note: suggested alternative: 'paused'
>           drawScreenSaver (display, elapsed, forceRedrawScreen);
>                                    ^ ~~~~~~
>                                    paused
> Clock: 280: 43: error: 'forceRedrawScreen' was not declared in this scope
>           drawScreenSaver (display, elapsed, forceRedrawScreen);
>                                             ^ ~~~~~~~~~~~~~~~~
> exit status 1
> 'elapsed' was not declared in this scope

Hoping that you can help me.
thanks a lot

QuinnFreedman, 2021-06-28:
Hi, thanks for giving it a try. Unfortunately, this is a known issue. The problem is that the Arduino Nano has a very limited amount of memory and just storing all the pixels for the screen takes up most of it. Enabling those options takes a few more bytes of memory which means that there is no room to allocate a buffer for the screen.

For now, if you want to enable some options, you will need to disable others. Disabling `CACHE_OUTPUTS_ENABLED` should be a free way to free up one extra byte, but you might have to disable swing or phase shift. I have been meaning to come back to this module to add a couple features and I may be able to optimize a few things slightly to make this better.

As for the screen saver issue, that sounds like a bug. I'll take a look at it as soon as I get a chance.

damwan, 2021-06-29:
Thank you for these explanations it is very clear.
I think I'll stick to the initial setup, I think the Swing and phase shifting will help me.
The clock works robustly and allowed me to synchronize, in addition to eurorack modules, pocket Operators without double trigger in slow tempo with the duty cycle adjustment.
Thanks for your work!

closed 2021-06-29 by damwan

QuinnFreedman, 2021-07-02:
The issue with screensavers should be fixed now. `SS_PULSE` seems to use a lot of memory though. Not sure why.

commit d37d8894 (2021-07-02): fixes #1