QuinnFreedman/modular #47: Add the current draw to each of the modules.

rl2939, 2024-08-18:
Add the current draw to each of the modules

It would be good to list how much current each module draws on the +12 volt, the -12 volt, and the 5 volt rails.
This would be useful information for those who DIY their power supply.

QuinnFreedman, 2024-08-18:
This is on my list to do. I just recently got the equipment to be able to measure this. I should be able to add these in the next month.

For now, you can assume the current draw is pretty low on all of them. There is no particular analog circuit that would burn a lot of power, so I'd guess they are all probably under 50mA. A large portion of that is going to be on the 5v rail for the digital modules.

QuinnFreedman, 2024-09-05:
It looks like my new power supply can't actually measure this on all 3 channels. I'll buy an actual ammeter. For now it looks like everything is in the 10mA or less neighborhood per rail except for Output which is ~40mA on both rails.

roagm, 2024-09-26:
I have measured the current draw for the modules that I have built. Here is the information in case you find it useful.

**Offset/Atten**
- 12V: 6 mA
- -12V: 6mA
- 5V: N/A

**Clock**
- 12V: N/A
- -12V: N/A
- 5V:
  - 30mA typical
  - 45 mA max (idle, all-white screen)

**Envelope**

- 12V: N/A
- -12V: 1 mA
- 5V 21mA

**Drift**

- 12V: 5 mA
- -12V: 5 mA
- 5V: 25 mA

roagm, 2024-09-29:
**RNG**

- 12V: 5mA
- -12V: 5mA
- 5V: Bias knob seems to affect current draw the most. Probably because it affects how many LEDs in the row of 7 light up
  - 22 mA with Bias knob all the way CCW (all row off)
  - 40 mA with Bias knob all the way CW (all row on)

commit 6de45c42 (2024-10-03): Add current draw and HP to modules (closes #47)
closed 2024-10-03 by QuinnFreedman