QuinnFreedman/modular #52: output module footprints

KeLaiFu, 2024-08-30:
I have the PCBs for this back from our friends at jlc and all is looking great apart from the footprint for the 6.35mm jack. I have ordered 3 different brands of the jacks and they all have a large flatter pin than the footprint. It's not a major problem as i have plenty of files, but it may be worth revising the footprints so more hardware will fit.

QuinnFreedman, 2024-08-30:
I designed it for ACJS-MN-3 jacks, which do fit nicely, but that was before I found the PJ-629HAN which are much cheaper. In addition to having wider flat pins, the PJ's also have a different pinout - left and right are reversed. I'd be interested to know what pinout your jacks are using -- I wonder if there is a standard.

I should probably update the design to use the PJ's. I'll officially add this to my to-do list

QuinnFreedman, 2024-08-30:
For my future reference -- I'm also seeing the [L11xxx](https://www.mouser.com/datasheet/2/393/110x_m110x_series_cd-2942927.pdf) series which offer both "snap in" and "wide" tails and follow the ACJM pinout, and the [NYS232](https://www.mouser.com/ProductDetail/REAN/NYS232?qs=%252B86TLfaev2%252B2HlyJ9Lqz%2FA%3D%3D) which has wide terminals and doesn't seem to list a pinout.

I guess I need to just pick one model and go with it

KeLaiFu, 2024-08-30:
I better check the pinout before I file anything then, I just picked a bunch that looked the same style on taobao, wasnt aware there could be different pinouts.