QuinnFreedman/modular #54: RNG: Values in loop not mutating

roagm, 2024-09-27:
The module always sends the same loop of CV values, regardless of the position of the Chance knob.

The Mutation Enabled LED does turn off when the Chance knob is at full CCW and it turns on when rotating it CW, but the values in the CV loop always remain the same. Turning the module off and on again changes the CV values in the loop, but the new sequence still does not mutate.

I have verified that my potentiometer is working correctly, and I have also made sure that the corresponding pin of the Arduino board is receiving the voltage changes. Additionally, I have tried using three different Arduino Nano boards, but that did not make any difference.

I am using the firmware hex file listed on the module's page

QuinnFreedman, 2024-09-27:
Oops! looks like the ability to actually update the buffer just got left out of the most recent firmware version. Surprised I didn't see that. Should be fixed momentarily.

closed 2024-09-27 by QuinnFreedman

dlakwi, 2024-09-28:
I can confirm that this fix resolves the issue of values not mutating. I built this module months ago but did not test it as thoroughly as roagm. Thank you roagm and Quinn.