QuinnFreedman/modular #55: RNG: 7 LEDs not all working

roagm, 2024-09-27:
Some LEDs of the group of 7 at the top of the module do not light up at all.

I have verified that the LEDs work, and I have tried with three different TLC5940NT ICs. With each IC, the LEDs that do not light up are different. I made sure that my three LED driver ICs work properly by testing them with another circuit. I also checked the continuity between the IC and the LEDs and found nothing unusual. Additionally, I have tried using three different Arduino Nano boards and the result is the same.

I am using the firmware hex file found in the module's page.

QuinnFreedman, 2024-09-27:
That is odd. I haven't seen that before.

Do the same LEDs stay off if you twiddle the Time/Bias knobs? Sounds more likely to be a hardware than software issue if the dead LEDs depends on the IC used but just want to make sure.

I know you tested for continuity but is it possible that it's a connection problem between the two boards? I was just working on a module where the stacking headers would sometimes break continuity depending on how you held them.

The TLC5940NT is honestly a huge pain to work with and requires very precise timing. Its conceivable that a different batch of ICs and/or Arduinos with different tolerances could have revealed a timing issue I wasn't experiencing. I'll try some other ICs and see if I find anything.

roagm, 2024-09-27:
The row of LEDs seems to work in the sense that I can see the lights scrolling from right to left depending on the Bias knob, and the length of the loop is shown when adjusting the Time knob. The thing is that some LEDs in that row never turn on. 

For instance, with one IC, the middle 3 LEDs never turn on, so I can see a light start scrolling from the right towards the left in time with the clock input. When the light gets to the 3 "dead" LEDs, they don't turn on, but when it is time for the light to move to the left of those 3 LEDs, it does and then it keeps scrolling to the next LED.
With that same IC, when I move the Time knob, I can only see when the loop has a length of 1, 2, or 32. I can still set it to 4, 8, or 16 by counting in my head how many steps I have turned the knob, I just can't see it indicated with the LEDs. The behavior is similar with the other TLC5940s that I have tried, but the "dead" LEDs are different with each IC.

I tried fiddling with the boards as you suggested, but I didn't notice any change in behavior.

I think it's more likely to be a timing issue as you mention rather than a hardware issue.

roagm, 2024-09-27:
I found the source of the problem. It is indeed a hardware issue.

The schematic for the back PCB has pin 19 of the TLC5940 (DCPRG) tied to GND. When I instead connect that pin to +5V, the LED row works properly.

Making that change in the schematic/Gerber files should probably be assessed

QuinnFreedman, 2024-09-27:
Wow, good detective work. I think I have gotten to the bottom of this issue.

The TLC5940 has the ability to add a per-LED correction factor to get a uniform brightness. It also gives you the ability to store those values to EEPROM (persistent storage). Pin 19 is DCPRG ("Dot correction programming"?). When DCPRG is grounded, it reads the correction values from EEPROM. When it is high, it uses the value that is currently in the dot correction register.

I never use this feature at all in the module, so I tied it low because all the chips I bought happened to have all high values in their EEPROM from the factory, so it effectively ignored this feature. But apparently they don't have to. Yours must have some random zeros in the storage from the factory.

Tying VPRG to 5V is probably the right way to go, but I want to double check that I'm not supposed to be writing some data to that register first and we don't get more undefined behavior. Alternatively, maybe I can fix this in firmware by just flashing the right values to the chips EEPROM on first time startup or something.

This is high on my priority now it should be not too bad of a fix one way the other

QuinnFreedman, 2024-09-27:
After looking at the datasheet some more, it looks like VPRG needs to be manipulated in order to write DC values, and I currently have it tied to ground, so I will need to change the PCB regardless. I will probably just tie DCPRG high, since that sounds easy. Also, the default DC register value is defined to be `0x3F` (the maximum value) so that should definitely work.

Thanks for your help tracking down this issue!

QuinnFreedman, 2024-10-24:
The version of the board with this fix was just merged into main after I finally got a chance to confirm it is all working

commit 11d3a754 (2024-09-28): Fix #55 and add decoupling capacitors
commit b4b80121 (2024-10-24): Fix #55 and add decoupling capacitors
closed 2024-10-24 by QuinnFreedman