QuinnFreedman/modular #67: O/A: VPP output difference between channels

extremenerble, 2025-04-19:
Hello Quinn,

First, thanks for all your work and publishing your designs. They're a real joy to build.

I built 2 O/A's and I'm seeing an issue replicated on both of them. The voltage output peak-to-peak is lower on channel 2 by about 1.0-1.3vpp for AC signals and 0.5-0.7vpp for DC signals. Also, full inversion on channel 2 gives you a slightly higher vpp (almost 0.5vpp for AC).

Some background/troubleshooting:
- I sourced all of the parts from Tayda, Mouser (LM324N's), or Aliexpress (pots)
- I've verified the cap and resistor values 
- I've swapped the op-amps around
- Reflowed everything twice more

Any input would be greatly appreciated!

QuinnFreedman, 2025-04-19:
Hi, that's definitely odd. I just did a little testing on my own copy to compare, and I found that my 2 channels behaved very consistently (within 20mV of each other).

I was testing a prototype v0.1 module with TL074's. It's possible I messed something up in the later revisions or that the LM324's are behaving differently. I can put a new module together and test it. My memory is a very minor bug fix was the only difference between those versions, though.

One big downside of the module is that the input is unbuffered (which is very silly. There are 2 unused op-amps in the design that I could easily use to buffer the input. I'll update that design soon). In my experience it behaves pretty precisely when the input is coming from a regulated source, but it gets very non-linear and weird when the input is coming from, e.g., a 100k pot bridging power and ground. I would be curious where the input you're testing with is coming from. If you can pass it through a buffered mult or use a regulated bench-top power supply as input that could remove a potential variable. Still, that doesn't explain why the second channel is consistently lower.

My only other idea is that R9 was incorrectly labelled in the schematic. It should be 10k (or the same as R16). It's listed correctly in the assembly instructions page, but it's listed as 100k in the schematic and interactive BoM. I fixed it in the project file but I guess I forgot to export and upload the fix. R9 *shouldn't* affect the output -- it just controls the LED brightness and the LM324's should have plenty of power to drive any value resistor there without sagging the output voltage at all, but if you have it as 100k it might be worth swapping down to 10k to see if that makes a difference.

extremenerble, 2025-04-19:
I did use the BOM in the building guide when I put them together, so the resistors are correct.

I'll test with running inputs through a buffered mult, and also swap in some TL074 and see if any of that makes a difference.

The inputs I'm testing with are Erica EDU VCO and a DIY Plaits, as well as triangle LFOs from a Zlob Skew-Fade and a CubuSynth 4VCLFO modules

Thanks for all of this info, it's really helpful!

extremenerble, 2025-04-19:
Ok so now I can't replicate the problem at all.

I piped in buffered signal and both modules were showing great on both channels. Then I went back to direct inputs and they're still behaving.

Might be dealing with a weird power issue. There are on my test skiff using a µZeus.

Thank you and looking forward to seeing your future designs!

closed 2025-04-19 by extremenerble